thijsbekke/OctoPrint-Pushover #1: Request: Add support for Job Started Notifcation

GForceWeb, 2016-08-24:
It would be great to also get a notification when a job starts (As in, bed warmed up and print commencing).

I like to watch the first couple minutes of a print to make sure the bed adhesion looks good but given my printer takes a while to heat up for ABS and it's in another room I often miss the start.

thijsbekke, 2016-08-24:
Thanks for the idea. Just give me a couple of days too develop it. You will hear from me soon.

thijsbekke, 2016-08-29:
It is a little bit harder than I original thought, the event `EVENT.PrintStarted` gets triggered when the print start and not when the print actually starts (The bed and hotend are on temp), I will keep working on it. There must be a way.

GForceWeb, 2016-08-30:
I'm not really familiar with how octoprint works and interacts with the 3D printer information so these might not be possible but you could look to possibly trigger the notification either:

- When the hotend/bed get up to the temps assigned (That would be just moments before the printer starts actually printing)
- When the first GCODE axis movements is fired?

thijsbekke, 2016-11-19:
After all I made up my mind and I am not developing this feature at this time. Recently I discovered that this plugin generates almost 4.000 notifications in a month. There is a limit on pushover after this limit you can't send notifications until you pay. And with this update there will be more notifications. In the most recent update I did make it possible for users too enter their own API key, maybe this will help too limit the number of notifications. My focus is too deliver a notification when a print is done or is failed. I hope you will understand. It is still a cool feature and maybe in the future.

Below is an image of the number of notifications.
[image]

closed 2016-11-19 by thijsbekke

CptanPanic, 2017-07-22:
Any chance this gets reopened?  Would like to get an alert when extruder gets to final temp.

thijsbekke, 2017-08-05:
Hey @CptanPanic, sorry for the late response, vacation and stuff. 

I understand the need for more notifications such as when the extruder or the bed gets to the final temp. But I can't implement too much notifications. Every month this API key is sending almost 10.000 (that is also the limit without paying for it) notifications. So some one has come up with a solution. See this issue; 
https://github.com/thijsbekke/OctoPrint-Pushover/issues/12 . When this is implemented we can look in too this .

Stay tuned.

CptanPanic, 2017-08-05:
Yeah you should definitely require people have their own API key.