thijsbekke/OctoPrint-Pushover #2: Customize pushover message

leftyfb, 2016-10-06:
I'd like to customize the message that comes from the pushover notification. At the moment they all start with "Thijs". I'd like it to say "3d printer" or "Octoprint" or something.

thijsbekke, 2016-10-07:
Will do, as soon as I get Back Home

thijsbekke, 2016-10-17:
@leftyfb can you try it again ? I released a new version but I never could of reproduce your bug. If it still exists can you post a screenshot ?

leftyfb, 2016-10-18:
[image]

It's till showing:   "Thijs: Print job...."

thijsbekke, 2016-11-16:
Found it, dumb mistake. Tonight you get a new update wit the fix. Sorry for the waiting.

thijsbekke, 2016-11-16:
This issue is fixed in the lastest release. Please update too 0.1.4. Thnx for your patience.

https://github.com/thijsbekke/OctoPrint-Pushover/tree/0.1.4

closed 2016-11-16 by thijsbekke