thijsbekke/OctoPrint-Pushover #5: Configure api_token?

neovanmatix, 2016-11-16:
Hey,

i would like to set the api_token, so i can change the pushover-channel/application.
It seems, that you hardcoded your api_token into init.py. I changed it to my own token, but it doesn't work (getting no pushover-msgs anymore :-/)

thijsbekke, 2016-11-16:
Hey, 

Normal behaviour is too only enter your user key from pushover. You can use my API code that is no problem. That is the way the provided API is designed. 

But thank you, because of your request I realized the wrong API key is in the code. (This will solve #2). I will fix this tonight and hope you don't want too change the hardcoded API key.

neovanmatix, 2016-11-16:
> I will fix this tonight and hope you don't want too change the hardcoded API key.
Of course, i would like to change the hardcoded API-Key to match my Pushover-Config (each Application has its own apikey, name & icon).

thijsbekke, 2016-11-16:
This issue is fixed in the lastest release. Please update too 0.1.4.

https://github.com/thijsbekke/OctoPrint-Pushover/tree/0.1.4

closed 2016-11-17 by thijsbekke