thijsbekke/OctoPrint-Pushover #7: install problems

analogmonster, 2016-12-09:
Hi, plugin not appearing in settings list after install, although is shown as installed in plugins manager.  Turn off plugin Octoprint asks to restart, reenable plugin and still doesn't show.  Un/reinstalled and it didn't help.  Other than a clean octoprint install, do you have any ideas?  Thanks

0.1.5 on 1.3.0 master.
pip 8.1.1 to 9.0.1 upgrade notification on install but no other errors.  I've tried on

thijsbekke, 2016-12-09:
Strange problem, I updated OctoPrint also to version 1.3.0 and beside some other problems I still see this plugin. But maybe we can locate the problem and fix it. 

You said the plugin manager said it is installed, can you check the log file ?
`cat ~/.octoprint/logs/octoprint.log|grep -a pushover`

Can you try too install the previous version of this plugin ? 
`pip install https://github.com/thijsbekke/OctoPrint-Pushover/archive/0.1.4.zip`

I hope one of these steps narrow down the problem.

analogmonster, 2016-12-09:
Don't worry, I've since discovered more problems so something went wrong when I did the updates. Just went to the plugin manager again and it now it says pip is missing which is a pretty good indication something really wrong is going on so I'm just going to do a fresh install. Thanks anyway

closed 2016-12-09 by analogmonster