thijsbekke/OctoPrint-Pushover #8: Request

coldfire3000, 2017-01-02:
Hi. Firstly, thanks for making this plugin for pushover, it's awesome!

I wondered if there was any way you could add a notification option, like you have for failed and completed, but for when a print is paused using the M0 command? Sometimes I add some pause code into my gcode so I can change colours and it works great. I make the printer beep when it pauses, so I know when to change colours, but sometimes I don't hear this from where I am. If I could get a pushover notification to say the pause has happened that would be amazing.

Thanks again for your work!

thijsbekke, 2017-01-02:
Hey, 

Thanks for the kind words. Please update too the latest version. It is now possible to send a Pushover notification when a ```M0``` command is received. And as bonus it will append any ```M70``` message to the notification, so you can remind yourself which colour you need to switch. 

For example 

```GCODE
G1 X109.071 Y96.268 E3.54401
G1 X109.186 Y97.500 E3.63927
M70 Sleep Message
M0 
G1 X109.186 Y102.500 E4.02408
G1 X108.789 Y104.770 E4.20140
```

closed 2017-01-02 by thijsbekke

coldfire3000, 2017-01-03:
That works great! Thanks for doing this so quickly, more amazing work!!