thijsbekke/OctoPrint-Pushover #10: [Request] Add support for event.waiting ?

kroyster, 2017-03-13:
Hey - I'm enjoying this plugin, but I'm not getting pause alerts for when I inject an M600 filament color change. I've tinkered with coding this myself, but no luck yet.  

I did notice in Octoprint's /OctoPrint/blob/master/src/octoprint/util/comm.py file, there's a gcodeToEvent section, that includes entries like "M226": Events.WAITING.  So this might be possible by adding an entry for "M600": Events.WAITING to the OctoPrint code, plus adding support for this plugin detecting these Waiting events.

coldfire3000, 2017-03-14:
In case it helps, I edit the Gcode manually when doing colour changes. If you use an M0 command to pause then you will get a notification and you can use M70 to send a specific message. I requested this a few months ago:
https://github.com/thijsbekke/OctoPrint-Pushover/issues/8

thijsbekke, 2017-03-20:
Hello @kroyster, You are right the command M600 isn't supported by either this plugin or OctoPrint, when I was testing it by adding M600  too the list of waiting commands, I saw it triggers a waiting event instead of a Pause event @coldfire3000 mentioned. 

```
gcodeToEvent = {
	# pause for user input
	"M226": Events.WAITING,
	"M0": Events.WAITING,
	"M1": Events.WAITING,
	"M600": Events.WAITING,
```

This plugin currently only supports the pause event (M0 and the M1) however I will create an update to support the Waiting event, and maybe I will try to support the M600 command (don't pin me on this). 

I will keep you posted.

@coldfire3000 Which command are you using M0 or M600 ?

coldfire3000, 2017-03-20:
I use the M0 and it works great for me :)

kroyster, 2017-03-20:
I use M600.  My understanding was that M600 automatically takes care of
raising and parking the hotend, retractions, locking the motors, etc.
whereas M0 requires additional gcode to accomplish all that.  If that is
not correct and they behave nearly the same, I suppose I could switch to
M0. (FWIW I'm using the Monoprice Mini Select, which I believe is Marlin
firmware.)

Keith Royster
keith@theRoysters.com

thijsbekke, 2017-03-21:
I implemented the wait event in this plugin. I recommend you use the M226 instead of the M600 or the M0. The M226 will (at least on my printer Prusa i3) raise and park the hotend, retract, make sound etc. 

And with the latest release it will also send a Pushover notification, remember you have to enable this through the settings. 

Can you please let me know if it doesn't work.

closed 2017-03-21 by thijsbekke

kroyster, 2017-03-21:
According to this, my Marlin firmware doesn't implement M226 as a pause, so
I may not be able to use that. But I will give it a try.

http://reprap.org/wiki/G-code#M226:_Gcode_Initiated_Pause

Keith Royster
keith@theRoysters.com

thijsbekke, 2017-03-21:
I hope it works otherwise someone has to ad the M600  to the util/comm.py file, maybe I will experiment with that later this week.

kroyster, 2017-03-22:
I updated the plugin and new settings, but M226 did not work for me, as it is not implemented (as a pause) in the Marlin firmware.  M600 is the only fully automated pause command that works for me, but I could not get it to trigger the Pushover notification. I did manually add a mapping for the M600 command to trigger a waiting state in the comms.py file, and rebooted octoprint, but it didn't seem to "take". So unfortunately I'm unable to confirm if this update works.  M0 pauses and triggers the Pushover alert, but doesn't retract/park my print head, so it's not useful to me for filament changes unless I can figure out how to add those commands AND return it to it's pre-pause location to resume the print.  Thank you though - I'm sure someone else will find this update helpful!

kroyster, 2017-03-22:
Correction - I just noticed that I edited the /src/ version of comm.py, but there are various other versions in oprint/lib/*, including compiled python libraries. So I'm guessing I need to edit one of the other files and/or compile it.  I tinker, but I'm not a developer, so it may take me a little digging to figure it out and give it another try.