thijsbekke/OctoPrint-Pushover #11: Option to change Octoprint Name in Pushover

rcallison, 2017-05-29:
I've got your plugin installed and working on a couple of installs of Octopi.  It is working great and very helpful.

Is there a way to have the Name that displays the message from Pushover be different?  I have 2 installations of Octopi and it would be helpful to me if I could distinguish which installation the messages are coming from.  I'm not sure if this is a function of Pushover or the plugin at this point.

Thank you for your work so far.

thijsbekke, 2017-05-31:
Hello, 

Thanks. According to the Pushover documentation it is possible. But not (yet) in this plugin. It is more an advanced option I think. There are too many configurations options already in my opinion. 

I am working on it.

coldfire3000, 2017-05-31:
This works fine for me... I have a separate pushover 'application' setup for each of my printers with their names as the pushover 'application' name. Then in Octoprint/printer 1 I use the API key from application 1 and printer2/octoprint2 I use application 2. So I have mine called 'Wanhao i3' and 'CR-10', then when I get the notification is says that as the first line. Works well for me :)

rcallison, 2017-05-31:
Thank you @thijsbekke for looking into it.  I'm still new to Pushover @coldfire3000 so I did not even realize this was possible.  I'll take a look and see if I can get it working.

rcallison, 2017-05-31:
@coldfire3000 thank you for the suggestion.  I've been playing around with Pushover and think I have this working as you suggest.  Works for me as a solutio.

thijsbekke, 2017-06-20:
See latest release. The printer name will now be included in the title of the notification.

closed 2017-06-20 by thijsbekke