thijsbekke/OctoPrint-Pushover #14: Send push message with webcam image

Nappyy, 2018-01-30:
Hi there,

since pushover 3.0 it supports also to send push messages with images.
It would be a cool feature if your plugin could send a webcam image of the finished print.

jbkuma, 2018-01-30:
I concur, I started looking through the code to see if I could implement it out on my own.  It is not immediately obvious to me where I would go about adding this.

thijsbekke, 2018-01-30:
Hey, 

That is a good idea ! I am already working on it, and it is almost done.

thijsbekke, 2018-01-31:
Almost there.. Only have to test it before releasing it.

https://github.com/thijsbekke/OctoPrint-Pushover/tree/image

thijsbekke, 2018-02-04:
See the new release (https://github.com/thijsbekke/OctoPrint-Pushover/releases/tag/0.2.0) 

Please let me know if it is not working !

closed 2018-02-04 by thijsbekke

Nappyy, 2018-02-07:
perfect, thanks!