thijsbekke/OctoPrint-Pushover #16: No connection

anfichtn, 2018-02-12:
The plugin tells me since the last update thats no internet connection could be established.... Why?
Using a console, i can reach pushover and any other site on the internet...

thijsbekke, 2018-02-12:
Thanks for opening this issue, I would look into it.

I have a couple of questions, Do you have a webcam and do you receive notifications ? 

You can try too uncheck the checkbox "Use the camera of Octoprint to send a image" in the setting maybe that works.

thijsbekke, 2018-03-20:
Closed due lack of response.

I did looked into it and the only code in OctoPrint that send suchs a message is the softwareupdate plugin. Can't help it, this plugin does not send such a message. I did however make a change that not every message is shown in the logs. Meaybe OctoPrint couldn't find your webcam url and thusfore gives this kind of message.

closed 2018-03-20 by thijsbekke