thijsbekke/OctoPrint-Pushover #17: [Request] Send an image every X% of progress

bagobones, 2018-02-18:
I currently use an external method to do this, but I would love for example to send a picture every 25% or so to be reminded to check on the status of the print and quickly see if it has failed.

Using my other method that currently uses homeassistant I send notifications at 5/25/50/75/99% completion.

bagobones, 2018-03-24:
Was rebuilding some of my automation and remembered this was still here. 

Any chance of this being implemented?

CptanPanic, 2018-04-29:
Yes you need to make it so we can add our own API key.

thijsbekke, 2019-01-28:
@CptanPanic In the last release (which I released a couple of minutes ago), you have the ability to obtain and enter your own API key. This opens the road to solving this issue and #25. stay tuned.

jefft8910, 2019-02-04:
> @CptanPanic In the last release (which I released a couple of minutes ago), you have the ability to obtain and enter your own API key. This opens the road to solving this issue and #25. stay tuned.

Guess I picked up this plugin at exactly the right time - this would prevent me from needing to set up a separate plugin with periodic emails to Pushover to address this.

markuskruse, 2019-03-11:
I will donate $50 if this request is implemented (sent through paypal or any other means that we agree on). All you others that like this, feel free to donate more than this. I will match the highest bidder (up to $200).

JasonMillward, 2019-03-12:
I'd like to throw my hand up for this one. I'm about 60% of the way through a merge request that will allow for sending every X minutes (and every Y minutes when under the first hour).

After that I'll see if it's possible to retrieve the progress percentage, if it is it should be easy enough to trigger on X%.

thijsbekke, 2019-03-14:
It is merged and will create a release soon. I want too ad some more events. 

Thnx

thijsbekke, 2019-03-19:
Added more events. I will test this tomorrow on an actual 3d printer. When that is a success it will be released, thanks for the patience.

bagobones, 2019-03-19:
Looking forward to testing it.

thijsbekke, 2019-03-20:
In the latest release it is now possible too receive an notification every X minutes.

closed 2019-03-20 by thijsbekke