thijsbekke/OctoPrint-Pushover #18: No notification on error

SebastianSchultz, 2018-02-26:
Hello,

I had a "thermal runaway" today andgot no notification.
"Print done" notification works.

thijsbekke, 2018-02-26:
Wow, that is not good. I will look into it.

bytefray, 2018-03-18:
Same here, just had a thermal runaway (false alarm, thankfully) and no notification was sent.

closed 2018-03-20 by thijsbekke

thijsbekke, 2018-03-20:
Hey,

I think I fixed this issue. However I can not test a thermal runnaway. So here is how I test it, when printing I disconnected my temperature sensor. And that gave me an error, and with the latest release a notification. 

Default those messages are high prio, you can change this behaviour under advanced options. Can you guys confirm this ?

bytefray, 2018-03-21:
Just installed the latest update and mistuned my PID settings so I could simulate the false alarm thermal runaway. This time, the plugin sent a high-priority Pushover alert:

"Error!! An error has occurred in the printer communication. Heating failed, system stopped! Heater_ID:0"

Looks like you fixed it. Thanks!!!

SebastianSchultz, 2018-04-03:
Here also :)
It just works for me :)

Thank you in advance!

commit cb44cbc7 (2018-03-20): Fixes #18

- #18 Send a notification on Error's and alerts when printing. Default those messages are high prio, you can change this behaviour under advanced options.
- #16 Not everthing is a exception
- make compatible with newer versions
- More documentation