thijsbekke/OctoPrint-Pushover #20: Notifications for "Alert event (M300)", cannot be turned off.

a1b0hph0b1a, 2018-03-28:
I needed to turn off the M300 noticications. But even though I set priority to "Off" plugin still sends a notification.

I use a few M300s in my manual g-code and I now get several alerts on phone that I don't need!

mmontminy, 2018-08-26:
I was struggling with the same issue. What fixed it for me was moving line 251 of __init__.py to line 250 indented to line 249.
```
if priority:
			payload["priority"] = priority
			self.event_message(payload)
```
Not sure if this is a typo, or intentional, but it appears to have fixed it for me. I don't know if it's created other issues though.

schnello, 2019-01-15:
@mmontminy Thanks a lot for this fix. Was already annoyed by the double message.

closed 2019-01-20 by thijsbekke

thijsbekke, 2019-01-20:
Fixed in the latest release. Thanks for the help !