thijsbekke/OctoPrint-Pushover #22: Not Sending Push Notifications

shmorgan, 2018-04-26:
As of yesterday, all push notification are no longer being delivered from any of my Octoprint Devices. I tested with the test option and watched the console on the pushover website and the Messages Delivered/Pending did not change.

shmorgan, 2018-04-26:
I am seeing this entry in the logs...

octoprint.log:2018-04-26 10:17:33,482 - octoprint.plugins.pushover - INFO - error while instantiating Pushover: HTTPSConnectionPool(host='api.pushover.net', port=443): Max retries exceeded with url: /1/users/validate.json (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x76cc2870>: Failed to establish a new connection: [Errno -2] Name or service not known',))
octoprint.log:|  Pushover (0.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_pushover

shmorgan, 2018-04-26:
Tested with there pyton test code and our keys, got this

send: 'POST /1/messages.json HTTP/1.1\r\nHost: api.pushover.net\r\nAccept-Encoding: identity\r\nContent-Length: 92\r\nContent-type: application/x-www-form-urlencoded\r\n\r\ntoken=apWqpdodabxA5Uw11rY4g4gC1Vbbrs&message=hello+world&user=u36baqgmw82bcqs5a6r9bmpjc2rh94'
reply: 'HTTP/1.1 429 Too Many Requests\r\n'

thijsbekke, 2018-04-26:
Hi, 

Thanks for contacting, I did not noticed it. This plugin uses an api key that can send 12,500 messages each month. And this plugin keeps getting more populair (which is a good thing) and therefore more messages are being sent. The reason why you can't send anymore messages is because the limit is reached. 

You can do 2 things, 
1. You wait, next month the counter is reset
2. Add your own api key (it is under advanced options) 

Also I recommend you remove your user key from the comment above (spam and stuff)

I will contact Pushover and hope they are kind enough the raise the limit (again). 

See the image below for a graph of the usage. 

[image]