thijsbekke/OctoPrint-Pushover #23: Installation failed

adrianmihalko, 2018-04-30:
I tried to install with Octoprint's Plugin Manager, but at the end I got error:

```
creating build/lib.linux-armv7l-2.7
creating build/lib.linux-armv7l-2.7/PIL
copying src/PIL/XVThumbImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/TarIO.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/SgiImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageWin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageTransform.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/__init__.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/Jpeg2KImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/FtexImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ExifTags.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageOps.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/PcfFontFile.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/GbrImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageFile.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/DdsImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/McIdasImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageDraw.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImagePalette.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageMath.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImtImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/PdfImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageDraw2.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/WebPImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageMorph.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/PyAccess.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/IcoImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/BmpImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/version.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/IcnsImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/MicImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/MpegImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/GimpPaletteFile.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/_tkinter_finder.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/IptcImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/WmfImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/features.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageSequence.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/GimpGradientFile.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageEnhance.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageQt.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/PcxImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/PcdImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/JpegImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/MspImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageFont.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageTk.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/PdfParser.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/SpiderImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/SunImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/XpmImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/GdImageFile.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageColor.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/EpsImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImagePath.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/PaletteFile.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/PpmImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ContainerIO.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageMode.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/WalImageFile.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageStat.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/FliImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/PSDraw.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/FontFile.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/BufrStubImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageFilter.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/_binary.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/FitsStubImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/DcxImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/TgaImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageCms.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/GifImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/PalmImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/BlpImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/BdfFontFile.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/PsdImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/FpxImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/MpoImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/OleFileIO.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/JpegPresets.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/XbmImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageGrab.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/PngImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/_util.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/PixarImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/Hdf5StubImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageChops.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/TiffTags.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/ImageShow.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/GribStubImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/Image.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/TiffImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
copying src/PIL/CurImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
running egg_info
writing src/Pillow.egg-info/PKG-INFO
writing top-level names to src/Pillow.egg-info/top_level.txt
writing dependency_links to src/Pillow.egg-info/dependency_links.txt
reading manifest file 'src/Pillow.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching '*.c'
warning: no files found matching '*.h'
warning: no files found matching '*.sh'
no previously-included directories found matching 'docs/_static'
warning: no previously-included files found matching '.appveyor.yml'
warning: no previously-included files found matching '.coveragerc'
warning: no previously-included files found matching '.codecov.yml'
warning: no previously-included files found matching '.editorconfig'
warning: no previously-included files found matching '.landscape.yaml'
warning: no previously-included files found matching '.travis'
warning: no previously-included files found matching '.travis/*'
warning: no previously-included files found matching 'build_children.sh'
warning: no previously-included files found matching 'tox.ini'
warning: no previously-included files matching '.git*' found anywhere in distribution
warning: no previously-included files matching '*.pyc' found anywhere in distribution
warning: no previously-included files matching '*.so' found anywhere in distribution
writing manifest file 'src/Pillow.egg-info/SOURCES.txt'
running build_ext
The headers or library files could not be found for jpeg,
a required dependency when compiling Pillow from source.
Please see the install instructions at:
https://pillow.readthedocs.io/en/latest/installation.html
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-UcPdCZ/pillow/setup.py", line 806, in <module>
raise RequiredDependencyException(msg)
__main__.RequiredDependencyException:
The headers or library files could not be found for jpeg,
a required dependency when compiling Pillow from source.
Please see the install instructions at:
https://pillow.readthedocs.io/en/latest/installation.html
----------------------------------------
Command "/home/octoprint/OctoPrint/venv/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-UcPdCZ/pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-OuJ5wm/install-record.txt --single-version-externally-managed --compile --install-headers /home/octoprint/OctoPrint/venv/include/site/python2.7/pillow" failed with error code 1 in /tmp/pip-install-UcPdCZ/pillow/
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output
```

thijsbekke, 2018-05-01:
Hi, 

Which version of OctoPrint are you using ?

adrianmihalko, 2018-05-01:
Hi,

I followed this tutorial:

http://deloarts.com/en/3d-printing/octoprint-on-orange-pi-zero/

+
Armbian Ubuntu server:
https://www.armbian.com/orange-pi-zero/

What should I install manually?

adrianmihalko, 2018-05-03:
I installed Octoprint image on a rPi, there is no problem with installation this plugin.

There is another problem, I set my user & api key. I test sending, but I get error in the log file (webinfo shows me that it was sent successfully):

```
2018-05-03 14:38:03,196 - octoprint.plugins.pushover - INFO - Connected to Pushover
2018-05-03 14:38:03,244 - octoprint.plugins.pushover - INFO - Could not send message: 'NoneType' object has no attribute 'read'
2018-05-03 14:39:00,605 - octoprint.plugins.pushover - INFO - Connected to Pushover
2018-05-03 14:39:57,303 - octoprint.plugins.pushover - INFO - Connected to Pushover
2018-05-03 14:39:57,348 - octoprint.plugins.pushover - INFO - Could not send message: 'NoneType' object has no attribute 'read'
2018-05-03 14:40:11,988 - octoprint.plugins.pushover - INFO - Connected to Pushover
2018-05-03 14:40:12,033 - octoprint.plugins.pushover - INFO - Could not send message: 'NoneType' object has no attribute 'read'
2018-05-03 14:40:28,887 - octoprint.plugins.pushover - INFO - Connected to Pushover
```

thijsbekke, 2018-05-03:
Can you try too uncheck the option "Use the camera of Octoprint to send a image" under the Pushover settings, and try again. If you have a camera attached please check the snapshot url, see if it is correct. 

[image]

I won't close this issue, I want too fix it, with a more clear error message.

rafalmag, 2019-11-05:
Hi @adrianmihalko,
I have similar issue - the installation crashes on pillow.
OrangePi PC Plus board with Armbian Linux 4.19.62-sunxi (based on Debian Stretch)
OctoPrint 1.3.12
How did you fix this?