thijsbekke/OctoPrint-Pushover #25: Feature Request

toddtmw, 2018-12-15:
Hi. I love this plug in. I can finally know when my prints finish.

One thing I would love to see is the option to motify when the print head and bed reach their target temperature. My printer for some reason does not wait for temp before starting, so I have to preheat it, but then sometimes I get distracted and forget to start the print for a while. It would be great to have a notificaiton when it reached temp.

Thank you for your awesome plug in. Please let me know your thoughts on whether this is possible.
Todd

ssvenn, 2019-01-20:
I would love more push event types too, especially a "print started" message. For some reason when I upload and start printing through Slic3r it sometimes doesn't actually start the print and it would be nice to have the extra confirmation.

markuskruse, 2019-01-21:
It would be good to have a notification after X minutes and then with Y minutes interval.
For instance after 5 minutes and then 60 minutes interval.

thijsbekke, 2019-03-20:
@markuskruse  @toddtmw In the latest release it is now possible too receive an notification every X minutes (thanks to @JasonMillward ) and too receive an event when the temperature is reached. 

@ssvenn Will create that one soon.

[image]

thijsbekke, 2019-07-30:
@ssvenn You mentioned you wanted a notification when your printer started printing. In the latest release you will find this functionality. 

https://github.com/thijsbekke/OctoPrint-Pushover/releases/tag/0.3.1

closed 2019-07-31 by thijsbekke