thijsbekke/OctoPrint-Pushover #30: Network Error

Simur, 2019-04-08:
Hello,

I've installed the plugin, he work normally.
But, after a server restart, the plugin don't work. 
the error is : 

> Sending failed: error while instantiating Pushover: HTTPSConnectionPool(host='api.pushover.net', port=443): Max retries exceeded with url: /1/users/validate.json (Caused by SSLError(SSLError("bad handshake: Error([('', 'osrandom_rand_bytes', 'getrandom() initialization failed.')],)",),))

Can you help me, please ? :D

thijsbekke, 2019-04-08:
Hi, 

Usually that means something is wrong at pushover side. Can't see anything wrong now and tested it at my own printer and it works. 

Maybe it was a hick up, can you please try it again ?

Simur, 2019-04-09:
Hello,
I have already tried to uninstall, and re-install again.
It works as long as I do not restart my Rpi.

Simur, 2019-04-25:
Hi =)
I still have the problem.
If it helps, here are the plugins that I have installed:
-Action Command Prompt
-Bed Visualizer
-Firmware Updater
-PSU Control
-Pushover
-TouchUI

Thanks for your help =)

thijsbekke, 2019-07-30:
Hi,

I dove into it, and unfortunately no luck finding a solution. 

For example see the link below, the mentioned the same error, and it had to do something with a linux kernel :/
https://gitlab.com/librespacefoundation/satnogs/satnogs-client-ansible/issues/29

It did help that you provided the plugins you installed. The plugins that you have installed are not connecting to the Internet. And it makes sense that you can not update OctoPrint because it needs a SSL connection too the Internet also. 

The problem is I can not fix something which I can not reproduce, did you tried to reinstall your Raspberry Pi ? 

https://octoprint.org/download/

Or maybe install another plugin which also needs a connection to the Internet ? 

Hope this helps.

closed 2019-07-30 by thijsbekke