thijsbekke/OctoPrint-Pushover #39: Command for saving picture to be used in job finished message

Wim-N, 2019-08-13:
Somebody else also asked about this in another way. When the job is finished the bed moves forward and out of view of the camera so I will always get a picture from an empty part of the bed. You adviced to use a 'Pause' code.
But is it possible to have a 'custom' G-code that will save the current image form the camera as a picture and that will be used in the job finished message. Then I can insert that code in the job end G-codes before the final move of the bed and the problem is solved.

ScottWell1, 2019-09-07:
Custom Gcodes are problematic since different printers implement their own sets of custom Gcodes.

The way I handle this issue....   The "end script" attached to the sliced file, coming from the slicer,  ends by moving the bed to center the camera view.  As far as Octoprint is considered, the job is complete at that point -- so the plugin fires, takes the picture, and sends the pushover notification.  After that, the Octoprint "After print job completes" script runs, and that's where the bed is moved to the print removal position.   So it's just a matter of removing the final "move bed to removal position" Gcode from the slicer script, and adding it to Octoprint's "Settings, Gcode Scripts, After print job completes" section.

Wim-N, 2019-09-09:
Thanks for the tip, I'll give it a try. But I might run into trouble when I take a gcode file prepared with this tip and put it on a SD-card for direct printing. In that case it will not have the full end commands in that file like diabling the stepper motors.
Another idea might be that when it 'sees' `the 'M104 S0 ;Turn-off hotend`' command it will take the picture. Then I only have the make sure that this command comes before the final movement of the bed. And that will not give any problems when printing from an SD-card.
Another thing is that all the comments in the gcode file is available for OctoPrint to read. So you might put the phrase '##take final picture##' in the comment of the end gcode and OctoPrint might capture a picture to be used.

ScottWell1, 2019-09-09:
Good point, I print exclusively via Octoprint and never use SD.   I'm certainly not against some other way to trigger the photo/pushover, as long as it doesn't conflict with any printer's implementation of Gcode.

But in the meantime, I think the only thing that needs to be removed from the slicer code is the final bed move to removal position.  Here's all I have in the "After print job completes" script of Octoprint:
  G90  ;absolute
  G1 Y180 F3000   ;move bed to object removal position
  M84  ;steppers off

For most printers, leaving the M84 in the ending Gcode from the slicer (executed just before the Octoprint "After print job completes" script) won't hurt anything.  The motors would automatically turn on again for the G1, then get turned off again.   Printing direct from SD would work, the only thing "missing" would be the final move to removal position.