thijsbekke/OctoPrint-Pushover #41: Excessive (and now CloudFlare DNS Blocked) API Requests to PushOver

HeyRobb, 2019-11-12:
In the last 9 hours, my PushOver addon has resolved api.pushover.net more than 33,000 times.  My home IP is now banned via CloudFlare... Any ideas why this might be happening? Running OctoPrint v1.3.11 and the latest version of this addon.  Any help would be greatly appreciated! (willing to upload logs as well if needed.. I've blocked api.pushover.net in PiHole DNS).

[https://imgur.com/a/TtJFwVU](url)

thijsbekke, 2019-11-12:
wow! That are a lot of requests, sorry to hear you are blocked. I have a couple of questions; 
do you have a valid api key filled in ? 
did you receive a notification every second ? 
did you receive any notification ?

If it is possible you can send the logs, maybe thats gives a clue what is going on.

HeyRobb, 2019-11-13:
I had my Pushover api keys for quite some time, no changes as far as I know to the plugin, or even OctoPrint to be honest that I can think of. I had gotten several notifications successfully from the plugin (with pictures) before things went haywire. I only received one notification per event (print 'first couple layers' notification and print complete notification, if I got any at all.. no repeats or floods.

Please let me know where to fetch the logs so I can upload them. Thank you!

thijsbekke, 2019-11-13:
I hope the log will not be too big with 33000  rules.

You can try the following, SSH into you octoprint machine/pi and enter the commands below. Ofcourse replace <date> with the date it happened.

```
cat ~/.octoprint/logs/octoprint.log.<date>|grep pushover > ~/log.txt
 ```
and then upload the log.txt in your home dir (~/log.txt)

I hope this will clarify some things.

HeyRobb, 2019-11-13:
As requested! Hope all of this helps.. looks like there was a lot of activity on the 10th as well. Thank you!

[octoprint.log.2019-11-10.txt](https://github.com/thijsbekke/OctoPrint-Pushover/files/3843944/octoprint.log.2019-11-10.txt)
[octoprint.log.2019-11-12.txt](https://github.com/thijsbekke/OctoPrint-Pushover/files/3843945/octoprint.log.2019-11-12.txt)

thijsbekke, 2019-11-14:
Thanks, 

I think I found the "problem", Your printer, the Prusa i3 MK3S has an auto level feature and when printing the Z axis has too move for compensating for the auto leveling feature. And it somehow send a notification / make contact too the API when doing that.

Bit tricky too reproduce, but will fix it.

HeyRobb, 2019-11-14:
No worries! Thank you for updating me on what is going on. If there is a way I can help test, please let me know!

kazibole, 2019-11-16:
A quick look at the logs show that this may be due to z-hop. PrusaSlicer defaults to 0.6 mm for z-hop and all of the z changes in the log are 0.6 mm.

HeyRobb, 2019-11-17:
Yep, can definitely confirm Z-hop is enabled @ 0.6mm (which is default for Prusa'a MK3S 0.6mm nozzle profile). Is that something that can be caught/excluded in a logic loop so notifications aren't sent as a result of the hop? It seems strange that I would be be the only one using that feature in a slicer. Can also test disabling that feature and printing if you'd like! For now I've turned off the 'after first couple layers' notification, and is working great!

thijsbekke, 2019-11-17:
Thanks for all the help, I think I found the issue (https://github.com/thijsbekke/OctoPrint-Pushover/commit/c9a5bcf3e5456bf79addc4b0e24631dcaf18b057), I am testing this fix tomorrow on my printer and will create a release then.

thijsbekke, 2019-11-19:
@HeyRobb I just released a fix which prevents a lot of messages to pushover, just update it through OctoPrint and all should be fine. Will you let me know when it is not fixed ? 
@kazibole Thanks

closed 2019-11-19 by thijsbekke