thijsbekke/OctoPrint-Pushover #48: Sends notification of first few layers complete after running a run off line

Syco54645, 2020-03-22:
Not sure what you want to call the line the printer draws at the start of a print. I have mine set to do a little purge on the left side. After it does this the plugin sends a notification that the first few layers are complete. Is there a way to account for this line?

thijsbekke, 2020-05-17:
Hi, 

The line around your print is called the brim. Does your z-axis move when it does the purge on the left side ? 

At least I did increase the value when to send the "first couple of layers are done" notification.

https://github.com/thijsbekke/OctoPrint-Pushover/pull/57/commits/ff5976d8fa3448c0bf579ee281e0ade5aa7ec1b8

It will be there with an update real soon

closed 2020-05-17 by thijsbekke

Syco54645, 2020-05-17:
This was the purge line that caused it to say the first few layers were done.

I actually disabled the plugin as I was getting several entries a second in the error log and it was causing a stalling when doing circles. I posted my log on reddit when I was having stalling issues, perhaps it will shine some light on the issue I was having.

https://www.reddit.com/r/octoprint/comments/g9oenz/any_solution_to_stutters_ender_3_with_abl_running/