thijsbekke/OctoPrint-Pushover #49: MMU Notifications

NHRA21, 2020-03-25:
I am looking for notification for when the Prusa MMU2 needs attention. It displays a message on the screen so I am guessing that there is a code that can be trapped.

fridgehead, 2020-10-14:
I've been looking into adding this myself as the MMU doesnt alert me when it needs kicking (which is pretty often) but it seems the printer doesnt issue any GCODE errors when problems are detected

I think the route to take is to watch for MMU messages using : https://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-gcode-received

The API doc warns about not doing any blocking requests in this handler, so I guess something as simple as looking for "Recv: OOOOOOOOOOOOOOOOOOOOOOOOOOOOOO failed." in the message should be ok but for actually alerting it'll need to push an event into a queue or something. 

If I get round to it this weekend I'll submit a PR if it works :)

Toriniasty, 2022-01-07:
I have written something for myself. If you want to give it a go:  
https://github.com/Toriniasty/OctoPrint-Pushover/commit/dac86fbd5cc7fabf4ac43667cae1ff5ca3f2ad09  
  
git clone plugin  
cd to directory  
activate octoprint virtualenv  
octoprint dev:plugin install (IIRC)