thijsbekke/OctoPrint-Pushover #53: BUG: Failure in newest update

Yeraze, 2020-05-13:
I updated to the latest version today (with Python3 support), and now most of my notifications are gone. I get % complete notifications, and "Print finished".. But bed temp and scheduled notifications don't come through.

The logs show this error:

```
2020-05-12 10:06:24,782 - octoprint.plugins.pushover - INFO - Got an event: DisplayLayerProgress_feedrateChanged Payload: {'changeFilamentTimeLeft': '-', 'totalHeightWithExtrusionFormatted': '9.9', 'feedrate': '7200', 'updateReason': 'fee
drateChanged', 'totalHeight': '9.89', 'totalLayer': '55', 'estimatedEndTime': '16:13', 'changeFilamentCount': 0, 'estimatedChangedFilamentTime': '-', 'printTimeLeftInSeconds': 22021, 'totalHeightFormatted': '9.9', 'currentHeight': '0.79',
 'printTimeLeft': u'6h7m1s', 'lastLayerDuration': '0h:15m:15s', 'averageLayerDurationInSeconds': '-', 'lastLayerDurationInSeconds': 915, 'changeFilamentTimeLeftInSeconds': 0, 'averageLayerDuration': '-', 'feedrateG1': '960', 'feedrateG0':
 '7200', 'totalHeightWithExtrusion': '9.888', 'fanspeed': '100%', 'progress': '7', 'currentHeightFormatted': '0.8', 'currentLayer': '3'}
2020-05-12 10:06:27,898 - octoprint.util.comm - ERROR - Error while processing hook pushover for phase sent and command M105:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3332, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_pushover/__init__.py", line 257, in sent_gcode
    self.check_schedule()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_pushover/__init__.py", line 236, in check_schedule
    self._settings.get(["events", "Scheduled", "message"]).format(elapsed_time=self.last_minute))
UnicodeEncodeError: 'ascii' codec can't encode character u'\u23f0' in position 42: ordinal not in range(128)
```

Yeraze, 2020-05-13:
Another one:
```
2020-05-12 10:25:01,776 - octoprint.plugins.pushover - INFO - Got an event: ZChange Payload: {u'new': 1.138, u'old': 0.963}
2020-05-12 10:25:01,780 - octoprint.plugin - ERROR - Error while calling plugin pushover
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_pushover/__init__.py", line 414, in on_event
    self._logger.debug("Payload message: %s " % str(payload["message"]))
UnicodeEncodeError: 'ascii' codec can't encode character u'\U0001f340' in position 32: ordinal not in range(128)
```

closed 2020-05-13 by thijsbekke

thijsbekke, 2020-05-13:
Hi,

Thanks for reporting, the issue is fixes in the latest release, (https://github.com/thijsbekke/OctoPrint-Pushover/releases/tag/0.3.5). When you have the time, can you test it please ?

PR #54 (2020-05-13): Bugfix
Fixes #53

Can not convert from unicode to text with str()
commit afa17636 (2020-05-13): Fix #53

Can not convert from unicode to text with str()