thijsbekke/OctoPrint-Pushover #58: Use camera to capture image not working

phoenix-down, 2020-07-02:
I have "Use camera of OctoPrint to send a image" option checked, but my push notification (iOS) does not contain any images. My OctoPrint webcam works fine otherwise.

[image]

thijsbekke, 2020-07-16:
Hi,

Please update to the latest release https://github.com/thijsbekke/OctoPrint-Pushover/releases/tag/0.3.8 and try again. If it still failes can you reopen this issue and post the log file. 

For more information how to find the log files see the OctoPrint community (https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299)

closed 2020-07-16 by thijsbekke

phoenix-down, 2020-07-18:
Thanks for suggesting the log. I figured it out:

`2020-07-18 02:23:37,437 - octoprint.plugins.pushover - DEBUG - Event triggered: PrintDone 
2020-07-18 02:23:37,439 - octoprint.plugins.pushover - DEBUG - Snapshot URL: /usr/bin/ffmpeg 
2020-07-18 02:23:37,439 - octoprint.plugins.pushover - INFO - Other error occurred while trying to get image: Invalid URL '/usr/bin/ffmpeg': No schema supplied. Perhaps you meant http:///usr/bin/ffmpeg? 
2020-07-18 02:23:37,440 - octoprint.plugins.pushover - INFO - Could not load image from url`

Turned out I only configured the streaming URL, not the snapshot URL. Once I configured the snapshot URL, it worked fine.