thijsbekke/OctoPrint-Pushover #63: no Pushover message with event PrintStarted

SenH, 2020-12-13:
I am not receiving a Pushover message upon the "Print Started" event.
When activating debug logging it appears this event triggers an 'invalid user' error?
```
2020-11-30 22:32:06,785 - octoprint.plugins.pushover - DEBUG - Got an event: PrintStarted, payload: {'name': 'CE3PRO_freezing-pixels-christmas-tree-a_02mm_pla_mk3s_.gcode', 'path': 'CE3PRO_freezing-pixels-christmas-tree-a_02mm_pla_mk3s_.gcode', 'origin': 'local', 'size': 2167685, 'owner': 'Sen', 'user': 'Sen'}
2020-11-30 22:32:06,786 - octoprint.plugins.pushover - DEBUG - Event triggered: PrintStarted 
2020-11-30 22:32:06,786 - octoprint.plugins.pushover - DEBUG - Snapshot URL: https://router.lan/nxwitness/snapshot.php 
2020-11-30 22:32:12,267 - octoprint.plugins.pushover - DEBUG - Response: b'{"user":"invalid","errors":["user identifier is not a valid user, group, or subscribed user key"],"status":0,"request":"b156f17a-a0f6-4a32-b357-b6ff68803c45"}'
```

Pushover messages do arrive for all other activated events (Startup, Shutdown, Print Done, Percent schedule notifications).
```
2020-11-30 23:34:17,668 - octoprint.plugins.pushover - DEBUG - Got an event: DisplayLayerProgress_heightChanged, payload: {'updateReason': 'heightChanged', 'totalLayer': '1026', 'currentLayer': '56', 'lastLayerDuration': '0h:00m:21s', 'lastLayerDurationInSeconds': 21, 'averageLayerDuration': '0h:00m:21s', 'averageLayerDurationInSeconds': 21, 'currentHeight': '11.47', 'currentHeightFormatted': '11.5', 'totalHeight': '205.3', 'totalHeightFormatted': '205.3', 'feedrate': '1200', 'feedrateG0': '9000', 'feedrateG1': '1200', 'fanspeed': '100%', 'progress': '10', 'printTimeLeft': '9h18m10s', 'printTimeLeftInSeconds': 33490, 'estimatedEndTime': '08:52', 'estimatedChangedFilamentTime': '-', 'changeFilamentTimeLeft': '-', 'changeFilamentTimeLeftInSeconds': 0, 'changeFilamentCount': 0}
2020-11-30 23:34:17,668 - octoprint.plugins.pushover - DEBUG - event: DisplayLayerProgress_heightChanged has an AttributeError {'updateReason': 'heightChanged', 'totalLayer': '1026', 'currentLayer': '56', 'lastLayerDuration': '0h:00m:21s', 'lastLayerDurationInSeconds': 21, 'averageLayerDuration': '0h:00m:21s', 'averageLayerDurationInSeconds': 21, 'currentHeight': '11.47', 'currentHeightFormatted': '11.5', 'totalHeight': '205.3', 'totalHeightFormatted': '205.3', 'feedrate': '1200', 'feedrateG0': '9000', 'feedrateG1': '1200', 'fanspeed': '100%', 'progress': '10', 'printTimeLeft': '9h18m10s', 'printTimeLeftInSeconds': 33490, 'estimatedEndTime': '08:52', 'estimatedChangedFilamentTime': '-', 'changeFilamentTimeLeft': '-', 'changeFilamentTimeLeftInSeconds': 0, 'changeFilamentCount': 0}
2020-11-30 23:34:17,753 - octoprint.plugins.pushover - DEBUG - Response: b'{"status":1,"request":"92d1b6a5-1f0e-49d7-8586-6f6bccbf0b0c"}'
...
2020-12-01 03:52:26,064 - octoprint.plugins.pushover - DEBUG - Got an event: PrintDone, payload: {'name': 'CE3PRO_freezing-pixels-christmas-tree-a_02mm_pla_mk3s_.gcode', 'path': 'CE3PRO_freezing-pixels-christmas-tree-a_02mm_pla_mk3s_.gcode', 'origin': 'local', 'size': 2167685, 'owner': 'Sen', 'time': 19219.36798565113}
2020-12-01 03:52:26,064 - octoprint.plugins.pushover - DEBUG - Event triggered: PrintDone 
2020-12-01 03:52:26,064 - octoprint.plugins.pushover - DEBUG - Snapshot URL: https://router.lan/nxwitness/snapshot.php 
2020-12-01 03:52:35,027 - octoprint.plugins.pushover - DEBUG - Response: b'{"status":1,"request":"d691ee0d-bacb-4c91-abef-8168e420c47f"}'
```

csigristgit, 2021-03-14:
I can confirm SenH's statement. it's exactly the same for me.

TheLion, 2021-09-03:
I can also confirm. No message when the print starts.

thijsbekke, 2021-09-23:
I will look into this as soon as I find the time for it. And that is first thing next week, Sorry

thijsbekke, 2021-09-28:
Aha, the printstarted event is deprecated, did not know that. So have too fix that

https://docs.octoprint.org/en/master/events/index.html#printing
```
PrintStarted
A print has started.

Payload:

name: the file’s name

path: the file’s path within its storage location

origin: the origin storage location of the file, either local or sdcard

size: the file’s size in bytes (if available)

owner: the user who started the print job (if available)

user: the user who started the print job (if available)

Deprecated since version 1.3.0:
file: the file’s full path on disk (local) or within its storage (sdcard). To be removed in 1.4.0.

filename: the file’s name. To be removed in 1.4.0.

Changed in version 1.4.0.
```

AlexVercammen, 2021-10-15:
Hello. Any update on the 2 fixes?

TheLion, 2021-10-15:
> 
> 
> Hello. Any update on the 2 fixes?

I can't say, I switched to a completely different notification system all-together.

TheLion, 2022-07-11:
I switched back to this plugin for notifying through Pushover and the Print Started notification is not send, so it is not fixed.

agibson2, 2024-10-30:
> I've just fixed the issue with the "PrintStarted" not sending SMS. Added the file to my FORK. [https://github.com/AlexVercammen/OctoPrint-Pushover](url)

Thanks for this fix.  Hopefully it gets integrated into this repo and pushed so that the original plugin in Plugin Manager works.