thijsbekke/OctoPrint-Pushover #66: Plugin interrupts startup....

ThomasCr, 2021-09-21:
Hi, with the plugin installed from repository and configured the webinterfaces from octoprint dont comes up.

I activated plugin debug and found this:

```
Sep 21 21:24:11 octopi octoprint[6409]: 2021-09-21 21:24:11,685 - octoprint.plugins.pushover - DEBUG - Got an event: ClientClosed, payload: {'remoteAddress': '192.168.99.50'}
Sep 21 21:24:11 octopi octoprint[6409]: 2021-09-21 21:24:11,687 - octoprint.plugins.pushover - DEBUG - event: ClientClosed has an AttributeError {'remoteAddress': '192.168.99.50'}
Sep 21 21:28:35 octopi octoprint[6409]: 2021-09-21 21:28:35,408 - octoprint.plugins.pushover - DEBUG - Got an event: UserLoggedIn, payload: {'username': 'thomas'}
Sep 21 21:28:35 octopi octoprint[6409]: 2021-09-21 21:28:35,408 - octoprint.plugins.pushover - DEBUG - event: UserLoggedIn has an AttributeError {'username': 'thomas'}
Sep 21 21:28:42 octopi octoprint[6409]: 2021-09-21 21:28:42,889 - octoprint.plugins.pushover - DEBUG - Got an event: ClientClosed, payload: {'remoteAddress': '192.168.99.50'}
Sep 21 21:28:42 octopi octoprint[6409]: 2021-09-21 21:28:42,890 - octoprint.plugins.pushover - DEBUG - event: ClientClosed has an AttributeError {'remoteAddress': '192.168.99.50'}
```

this messages loops multiple times ...

Any idea what could me wrong?

ThomasCr, 2021-09-21:
192.168.99.50 is the client ip who wants to access the webinterface.

ThomasCr, 2021-09-23:
I think I know why this happend.... your code lacks a timeout definition on the web requests.

> When you make an inline request to an external service, your system will need to wait upon the response before moving on. **If your application waits too long** for that response, requests to your service could back up, **your user experience could suffer, or your background jobs could hang**.
> 
> **By default, requests will wait indefinitely** on the response, so you should almost always specify a timeout duration to prevent these things from happening.

Source: https://realpython.com/python-requests/#timeouts

as seen on those code-lines:

https://github.com/thijsbekke/OctoPrint-Pushover/blob/1166559d5ee480e3b2b56d492c984838b4585408/octoprint_pushover/__init__.py#L112
https://github.com/thijsbekke/OctoPrint-Pushover/blob/1166559d5ee480e3b2b56d492c984838b4585408/octoprint_pushover/__init__.py#L494

my problem was a faulty ipv6 setup - but this should to end in a not working octoprint ;-)

AlexVercammen, 2021-09-23:
Could this lack of timeouts being the reason for the issue #63 "no Pushover message with event PrintStarted"?[https://github.com/thijsbekke/OctoPrint-Pushover/issues/63](url)

thijsbekke, 2021-09-23:
I will look into this as soon as I find the time for it. And that is first thing next week. I think @ThomasCr is right here, a timeout definition is missing.

thijsbekke, 2021-09-28:
This issue is fixed but I wait with the release when https://github.com/thijsbekke/OctoPrint-Pushover/issues/63 is also fixed.